Skorpi08/Nextion4Marlin #5: Hotend fan not coming on

klack, 2020-08-24:
#define E1_AUTO_FAN_PIN FAN2_PIN
#define E0_AUTO_FAN_PIN FAN2_PIN

should be

#define E0_AUTO_FAN_PIN FAN1_PIN
#define E1_AUTO_FAN_PIN FAN1_PIN

Thank you for your work on this!

Skorpi08, 2020-08-24:
i have my own pins_TENLOG_D3_HERO.h  see Marlin_2.0.6- HICTOP.zip

klack, 2020-08-24:
I forgot to specify.  This is in configuration_adv.h in the auto cooling section.

klack, 2020-08-24:
Would you consider unzipping the folder and committing it?  We could submit these as pull requests making less work for you.

Skorpi08, 2020-08-24:
No.
What you need is only configuration Files.
Maybe in next version.

Btw you can use LCD12864 with my pins_TENLOG_D3_HERO.h in Marlin_2.0.6- HICTOP.zip
https://cdn.discordapp.com/attachments/593010746431111196/744715282567921766/IMG_20200817_023129.jpg

klack, 2020-08-24:
Ah my mistake.  I got it mixed up with the pin file that was recently added to Marlin.

closed 2020-08-24 by klack