Skorpi08/Nextion4Marlin #7: DEFAULT_DUAL_X_CARRIAGE_MODE should be DXC_AUTO_PARK_MODE

klack, 2020-08-24:
This is more of a suggestion.  Users who have not previously setup full control options in their slicer will have difficulty (like me :)  

```DEFAULT_DUAL_X_CARRIAGE_MODE DXC_AUTO_PARK_MODE```

Located at:
```Nextion-for-Marlin-2/Marlin_2.0.6- HICTOP/Marlin_2.0.6- HICTOP/Marlin/Configuration_adv.h:612```

ccomincini, 2020-08-30:
Hey @klack, I had some difficulty too in understanding that. Thank-you. 
On mine Tenlog D3 Pro, when I have selected the second extruder as the active one, if I home X-axis the second extruder park itself and after that goes straight to X0 Y0.
Do you know which option lays the extruder parked? 
If I change to the left extruder and home X-Axis (or All axes) the two nozzles stay parked at the right place.

klack, 2020-08-30:
I still have this issue too.  I have to issue T0 command before homing.

ccomincini, 2020-08-31:
> I still have this issue too. I have to issue T0 command before homing.

Hi Klack, I've found a lot of discussion on this function that put newly selected extruder in the same position of the previous one as you issue a Tool Change command. Actually if you issue a "T1 S1" instead than a "T1" command, the second extruder does not move. 
Moreover in the configuration_adv.h file there's a way to make this behavior the default: it involves the   `#define TOOLCHANGE_NO_RETURN`  but when I've tried to enable the option, the auto park mode has stopped to work.