Skorpi08/Nextion4Marlin #8: hello where is the config for nextion to modificate some option ?

labotecno, 2020-08-25:
hello thank you very much for this firmware is very good work ! 
can you tell me where is the config file for nextion screen ? i wand to add some feature 
i dont see any nextion file on your firmware marlin 
thank you

Skorpi08, 2020-08-25:
What you need is only Configuration.h and Configuration_adv.h

ccomincini, 2020-08-29:
Hi, thank you for sharing this wonderful work.
I've tried to open the Hictop D3 Hero.HMI  file but the editor refuses to open it. Neither the Chinese TJC nor the official Nextion editor.
The other file in the repository, iRics.HMI, opens without errors. I'd like to have an iRics similar menu on my Tenlog... in your youtube video, you say that you are using that TFT configuration for your Hictop D3. Is it possible to use that? I'm a rookie in firmware mod and I'd really appreciate your help.

labotecno, 2020-08-30:
i still don t understand how  there are nothing about lib for nextion in you firmware , can you tell how you do it?

Skorpi08, 2020-08-30:
just like BTT:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware

labotecno, 2020-08-30:
> 
> 
> just like BTT:
> https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware

or soo every thing is inside the hmi file? firmware for nextion... ?

Skorpi08, 2020-08-30:
yes

ccomincini, 2020-08-30:
Yes but the HMI file must be opened with the Chinese version of the editor. The HMI nextion editor refuses to open the Hictop HMI file but it opens the iRics one.
If you cannot understand Chinese language, you can download the Chinese version from here: [http://wiki.tjc1688.com/doku.php?id=2.%E4%B8%8A%E4%BD%8D%E8%BD%AF%E4%BB%B6%E4%B8%8B%E8%BD%BD:2.%E5%8E%86%E5%8F%B2%E7%89%88%E6%9C%AC%E4%B8%8B%E8%BD%BD](http://wiki.tjc1688.com/doku.php?id=2.%E4%B8%8A%E4%BD%8D%E8%BD%AF%E4%BB%B6%E4%B8%8B%E8%BD%BD:2.%E5%8E%86%E5%8F%B2%E7%89%88%E6%9C%AC%E4%B8%8B%E8%BD%BD) and compare with the English Nextion version to understand commands, as the two interfaces are identical).

klack, 2020-09-01:
> Yes but the HMI file must be opened with the Chinese version of the editor. The HMI nextion editor refuses to open the Hictop HMI file but it opens the iRics one.
> If you cannot understand Chinese [...]

This might be of use:
https://unofficialnextion.com/t/howto-use-tjc-editor-v0-55-for-tjc-panels-in-english/39

fsadak, 2020-09-30:
Hi,
First of all, thank you for your work and sharing.

When I tried your work, I saw that information was sent from the screen to the printer, but not from the printer to the screen. Is it necessary to make changes in Marlin software except HMI file? If so, I ask you to share what kind of changes you have made in the Marlin version you shared. Thanks...

Voodoobrew101, 2021-02-28:
Could we just flash the BTT firmware to the screen?