Skorpi08/Nextion4Marlin #12: tenlog tl-d3 pro 2208

Voodoobrew101, 2021-02-24:
I am not sure where I could ask and get results. So far you have done more then any one else I have seen.
I have a Tenlog tl-d3 pro which as you probably know is like the Hictop D3 Hero.
I would love to have it run Marlon 2.0 with 2208 stepper drivers and magnetic stops. 
The display is serial so I think it should work.
I could not compile your firmware in ardunioIDE or VSCode even without changing any values.
I would appreciate any help you could offer.

Voodoobrew101, 2021-02-24:
I may have found the problem. There are issues with the newer versions of VSCode or PI.o that are incompatable with the previous version of Marlin. So I am coppying your settings into Marlin 2.0.7.2, it appears to be working.

Voodoobrew101, 2021-02-28:
My copying changes from Configuration.h and Configuration_adv.h and didn't work. display just show revolving wheel, and cant connect from usb. I did not flash the display, is that expected behavior with wrong display firmware?

tonnic, 2021-05-22:
Any updates here on this?

Voodoobrew101, 2021-05-23:
I decided to go with the skr 1.4 turbo board with the motor expansion and tft 4.3 screen since whenever the usb was conected the screen kept trying too boot up. The newest marlin firmware has a way to include the motor expansion board but i dont have it fully conected yet. You might need to use an older version of marlin to get this to work and use the linux command line stuff in windows to get it to compile. That was too much for me to figure out.

tonnic, 2021-05-23:
> I decided to go with the skr 1.4 turbo board with the motor expansion and tft 4.3 screen since whenever the usb was conected the screen kept trying too boot up. The newest marlin firmware has a way [...]

What about the connectors for the machine?  All of their connectors are proprietary, so what did you do there?